LONGER3D/Marlin2.0-lgt #1: [FR] (feature request title)

FulvioSpelta, 2021-05-22:
Hi all, I'd like to have Marlin 2 on my alfawise U30 Pro. I'm new to fw and 3d printer world so maybe that i'm a bit confused.

I kindly ask:
1) is this fw the "reference" for U30 Pro ?
2) Is the released bin files suitable for update the U30 Pro printer ?
3) what "environment" should be compiled to contribute ?

Using master branch both STM32F103VE_longer and mega2560 environments generate blocking compiler errors.

Thanks a lot

LONGER3D, 2021-05-31:
We are sorry to bring trouble you. We will update document(README) to explain it clearly

FulvioSpelta, 2021-05-31:
Thanks a lot, i'll wait for the updated document.
I close the issue that is no longer useful now.

closed 2021-05-31 by FulvioSpelta