LONGER3D/Marlin2.0-lgt #3: [BUG] Will Not Compile with Longer Board or Mega 2560 selected.

crypticinteractive, 2021-09-03:
This will not compile. Issue was opened in May with the feedback that read me would be updated with info to build. 

Trying to build for LK5 Pro since you did not release a build that supports this model. 

Compiling .pio\build\STM32F103VE_longer\libb10\U8glib-HAL\clib\u8g_com_atmega_hw_spi.c.o
In file included from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:58:0,
                 from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from .pio\libdeps\STM32F103VE_longer\U8glib-HAL\src\clib\u8g_com_arduino_fast_parallel.c:69:
C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/ext_interrupts.h:90:6: error: conflicting types for 'attachInterrupt'
 void attachInterrupt(uint8 pin, voidArgumentFuncPtr handler, void *arg,
      ^~~~~~~~~~~~~~~
C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/ext_interrupts.h:69:6: note: previous declaration of 'attachInterrupt' was here        
 void attachInterrupt(uint8 pin, voidFuncPtr handler, ExtIntTriggerMode mode);
      ^~~~~~~~~~~~~~~
In file included from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:60:0,
                 from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from .pio\libdeps\STM32F103VE_longer\U8glib-HAL\src\clib\u8g_com_arduino_fast_parallel.c:69:
C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish_math.h:52:6: error: conflicting types for 'random'
 long random(long max);
      ^~~~~~
In file included from c:\users\cript\.platformio\packages\toolchain-gccarmnoneeabi@1.70201.0\arm-none-eabi\include\stdlib.h:11:0,
                 from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:41,
                 from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from .pio\libdeps\STM32F103VE_longer\U8glib-HAL\src\clib\u8g_com_arduino_fast_parallel.c:69:
c:\users\cript\.platformio\packages\toolchain-gccarmnoneeabi@1.70201.0\arm-none-eabi\include\stdlib.h:255:6: note: previous declaration of 'random' was here
 long _EXFUN(random,(_VOID));
      ^
In file included from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:60:0,
                 from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from .pio\libdeps\STM32F103VE_longer\U8glib-HAL\src\clib\u8g_com_arduino_fast_parallel.c:69:
C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish_math.h:61:6: error: conflicting types for 'random'
 long random(long min, long max);
      ^~~~~~
In file included from c:\users\cript\.platformio\packages\toolchain-gccarmnoneeabi@1.70201.0\arm-none-eabi\include\stdlib.h:11:0,
                 from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:41,
                 from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from .pio\libdeps\STM32F103VE_longer\U8glib-HAL\src\clib\u8g_com_arduino_fast_parallel.c:69:
c:\users\cript\.platformio\packages\toolchain-gccarmnoneeabi@1.70201.0\arm-none-eabi\include\stdlib.h:255:6: note: previous declaration of 'random' was here
 long _EXFUN(random,(_VOID));
      ^
In file included from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:60:0,
                 from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from .pio\libdeps\STM32F103VE_longer\U8glib-HAL\src\clib\u8g_com_arduino_fast_parallel.c:69:
C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish_math.h:181:17: error: conflicting types for 'makeWord'
 extern uint16_t makeWord( uint8_t h, uint8_t l ) ;
                 ^~~~~~~~
C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish_math.h:180:17: note: previous declaration of 'makeWord' was here
 extern uint16_t makeWord( uint16_t w ) ;
                 ^~~~~~~~
In file included from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:63:0,
                 from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from .pio\libdeps\STM32F103VE_longer\U8glib-HAL\src\clib\u8g_com_arduino_fast_parallel.c:69:
C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wiring_pulse.h:34:72: error: expected ';', ',' or ')' before '=' token
 uint32_t pulseIn( uint32_t ulPin, uint32_t ulState, uint32_t ulTimeout = 1000000L ) ; 
                                                                        ^
compilation terminated due to -fmax-errors=5.
In file included from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:58:0,
                 from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from .pio\libdeps\STM32F103VE_longer\U8glib-HAL\src\clib\u8g_com_arduino_common.c:46:
C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/ext_interrupts.h:90:6: error: conflicting types for 'attachInterrupt'
 void attachInterrupt(uint8 pin, voidArgumentFuncPtr handler, void *arg,
      ^~~~~~~~~~~~~~~
C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/ext_interrupts.h:69:6: note: previous declaration of 'attachInterrupt' was here        
 void attachInterrupt(uint8 pin, voidFuncPtr handler, ExtIntTriggerMode mode);
      ^~~~~~~~~~~~~~~
*** [.pio\build\STM32F103VE_longer\libb10\U8glib-HAL\clib\u8g_com_arduino_fast_parallel.c.o] Error 1
In file included from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:60:0,
                 from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from .pio\libdeps\STM32F103VE_longer\U8glib-HAL\src\clib\u8g_com_arduino_common.c:46:
C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish_math.h:52:6: error: conflicting types for 'random'
[... 435 lines cut ...]
In file included from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:63:0,
                 from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from .pio\libdeps\STM32F103VE_longer\U8glib-HAL\src\clib\u8g_com_arduino_st7920_custom.c:58:
C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wiring_pulse.h:34:72: error: expected ';', ',' or ')' before '=' token
 uint32_t pulseIn( uint32_t ulPin, uint32_t ulState, uint32_t ulTimeout = 1000000L ) ; 
                                                                        ^
In file included from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:63:0,
                 from C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from .pio\libdeps\STM32F103VE_longer\U8glib-HAL\src\clib\u8g_com_arduino_t6963.c:71:
C:\Users\cript\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wiring_pulse.h:34:72: error: expected ';', ',' or ')' before '=' token
 uint32_t pulseIn( uint32_t ulPin, uint32_t ulState, uint32_t ulTimeout = 1000000L ) ; 
                                                                        ^
*** [.pio\build\STM32F103VE_longer\libb10\U8glib-HAL\clib\u8g_com_arduino_std_sw_spi.c.o] Error 1
*** [.pio\build\STM32F103VE_longer\libb10\U8glib-HAL\clib\u8g_com_arduino_st7920_spi.c.o] Error 1
*** [.pio\build\STM32F103VE_longer\libb10\U8glib-HAL\clib\u8g_com_arduino_st7920_hw_spi.c.o] Error 1
*** [.pio\build\STM32F103VE_longer\libb10\U8glib-HAL\clib\u8g_com_arduino_sw_spi.c.o] Error 1
compilation terminated due to -fmax-errors=5.
compilation terminated due to -fmax-errors=5.
*** [.pio\build\STM32F103VE_longer\libb10\U8glib-HAL\clib\u8g_com_arduino_st7920_custom.c.o] Error 1
*** [.pio\build\STM32F103VE_longer\libb10\U8glib-HAL\clib\u8g_com_arduino_t6963.c.o] Error 1
============================= [FAILED] Took 8.79 seconds =============================

Environment         Status    Duration
------------------  --------  ------------
STM32F103VE_longer  FAILED    00:00:08.788

crypticinteractive, 2021-09-03:
Figured it out. 
Create new Project in PlatformIO. Select STM32F103VE Generic Board. This will fix errors on build.

Baddflash, 2021-09-05:
I agree, this has no reference to the pro models and is not pinned for the 8-bit boards.

LONGER3D, 2021-09-15:
please checkout to LKxPro branch, default master branch is for LKx 32bit board

crypticinteractive, 2021-09-15:
> please checkout to LKxPro branch, default master branch is for LKx 32bit board
> 
> 

Where do we find such board? I see no mention of this board anywhere on site. If available I would like to purchase if this is all that is needed.

LONGER3D, 2021-09-16:
below is the LONGER offical purchase link to 32bit board. alternative for  LKxPro 8bit board. 
https://www.longer3d.com/collections/fdm-mainboard/products/mainboard-of-longer-lk4-lk1-u20-u20-plus-u30

LONGER3D, 2021-09-16:
> > please checkout to LKxPro branch, default master branch is for LKx 32bit board
> 
> Where do we find such board? I see no mention of this board anywhere on site. If available I would like to purchase if this is all that is needed.

below is the LONGER offical purchase link to 32bit board. alternative for LKxPro 8bit board.
https://www.longer3d.com/collections/fdm-mainboard/products/mainboard-of-longer-lk4-lk1-u20-u20-plus-u30

crypticinteractive, 2021-09-16:
> below is the LONGER offical purchase link to 32bit board. alternative for LKxPro 8bit board.
>
> https://www.longer3d.com/collections/fdm-mainboard/products/mainboard-of-longer-lk4-lk1-u20-u20-plus-u30

Awesome. You have been helpful. Another quick question, is the LK5 Pro being abandoned? I ask as there is little to no reference to the printer in git and the board link you provided has no firmware build for the LK5 Pro.

LONGER3D, 2021-09-16:
> Awesome. You have been helpful. Another quick question, is the LK5 Pro being abandoned? I ask as there is little to no reference to the printer in git and the board link you provided has no firmware [...]

but not foget to match specific touch screen. we recommend you to get all the stuff you need in offical alipress store. below is the link:
[main board](https://www.aliexpress.com/item/32944556459.html)
[Touchscreen ](https://it.aliexpress.com/item/32946963998.html)

LONGER3D, 2021-09-16:
We are sorry that the README did not describe clearly how to complile firmware for LKxPro series. We will touch up README later.

Baddflash, 2021-09-16:
Unless that board has had the stepper drivers upgraded , they are not the silent type. 
To lose the silent, socketed steppers drivers to only gain the 32-bit processor is a major step back.
32-bit brings little to FDM 3d printers at all, and unless the software is specifically designed to take advantage of the more powerful 32-bit chip you gain absolutely nothing over 8-bit system.

I would keep your mainboard, and just use the correct firmware.  Or purchase a 3rd party board that has the silent/socketed 
steppers and a 32-bit chip.

LONGER3D, 2021-10-13:
If tjhe issue is solved. I will close this post