LONGER3D/Marlin2.0-lgt #4: [BUG] (short description)

ThomasCr, 2021-09-20:
[image]

Now, for LK5 Pro - I have to install LK4 FW or what?

ThomasCr, 2021-09-25:
After more reading I know LK4 cant be compatible with LK5_Pro...

I have seen, that there is a Branch for LKxPro..... I tested to build the firmware by myself with this branch.

But I fail on the bed BLTouch build.... I have set all params for LK4_Pro_BLTOUCH from Longer Marlin 1.1.9 repository and relevant things. I can also follow the procedure to set Z... but later when I try to start G29 I get a error.

The z axis moves up to high and starts there a probe. but is not moving down.

```
Printer is now online.
echo:Marlin V1.0-Marlin2.x
echo: Last Updated: 2021-06-18 | Author: Longer3D, Hobi, tpruvot
echo:Compiled: Sep 25 2021
echo: Free Memory: 2503  PlannerBufferBytes: 1200
echo:V76 stored settings retrieved (656 bytes; crc 62074)
echo:  G21    ; Units in mm (mm)
echo:; Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:; Steps per unit:
echo: M92 X80.00 Y80.00 Z400.00 E98.00
echo:; Maximum feedrates (units/s):
echo:  M203 X200.00 Y200.00 Z5.00 E25.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X200.00 Y200.00 Z100.00 E3000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P200.00 R500.00 T200.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.40 E5.00
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Auto Bed Leveling:
echo:  M420 S0 Z5.00
echo:  G29 W I0 J0 Z0.00000
echo:  G29 W I1 J0 Z0.00000
echo:  G29 W I2 J0 Z0.00000
echo:  G29 W I3 J0 Z0.00000
echo:  G29 W I0 J1 Z0.00000
echo:  G29 W I1 J1 Z0.00000
echo:  G29 W I2 J1 Z0.00000
echo:  G29 W I3 J1 Z0.00000
echo:  G29 W I0 J2 Z0.00000
echo:  G29 W I1 J2 Z0.00000
echo:  G29 W I2 J2 Z0.00000
echo:  G29 W I3 J2 Z0.00000
echo:  G29 W I0 J3 Z0.00000
echo:  G29 W I1 J3 Z0.00000
echo:  G29 W I2 J3 Z0.00000
echo:  G29 W I3 J3 Z0.00000
echo:; PID settings:
echo:  M301 P17.22 I1.00 D74.22
echo:  M304 P841.68 I152.12 D1164.25
echo:; Power-Loss Recovery:
echo:  M413 S1
echo:; Z-Probe Offset (mm):
echo:  M851 X-36.00 Y-4.00 Z-6.90
echo:; Filament runout sensor:
echo:  M412 S1
echo:SD init fail
dw: begin
Touch Screen: unknown, use default DWIN T5
echo:SD init fail
dw: go home page
>>> G28
SENDING:G28
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
[... 251 lines cut ...]
SENDING:G28 Z0
echo:busy: processing
echo:busy: processing
>>> M280 P0 S160
SENDING:M280 P0 S160
>>> G4 P100
SENDING:G4 P100
>>> G29
SENDING:G29
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
Error:!! STOP called because of BLTouch error - restart with M999
[ERROR] Error:!! STOP called because of BLTouch error - restart with M999

Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
[ERROR] Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)

Error:Probing Failed
[ERROR] Error:Probing Failed

```

ThomasCr, 2021-09-25:
```
diff --git "a/C:\\Users\\Thomas\\Downloads\\Marlin2.0-lgt-support-LKxPro.orig\\Marlin\\Configuration.h" "b/C:\\Users\\Thomas\\Downloads\\Marlin2.0-lgt-support-LKxPro\\Marlin\\Configuration.h"
index fcfb649..e0dd36c 100644
--- "a/C:\\Users\\Thomas\\Downloads\\Marlin2.0-lgt-support-LKxPro.orig\\Marlin\\Configuration.h"
+++ "b/C:\\Users\\Thomas\\Downloads\\Marlin2.0-lgt-support-LKxPro\\Marlin\\Configuration.h"
@@ -99,8 +99,14 @@
   #define LGT_LCD_DW    // DWIN 4.3 inch LCD serial touch screen
 #endif
 
-#define LK4_PRO
-// #define LK5_PRO
+//#define LK4_PRO
+
+#define LK5_PRO
+
+#ifdef LK5_PRO
+  #define LK4_Pro
+//  #define LK5
+#endif
 
 #define SHORT_BUILD_VERSION       "V1.0-Marlin2.x"
 #define DEFAULT_MACHINE_NAME      "LONGER 3D Printer"   // override by  CUSTOM_MACHINE_NAME if any
@@ -1004,7 +1010,7 @@
 /**
  * The BLTouch probe uses a Hall effect sensor and emulates a servo.
  */
-// #define BLTOUCH
+#define BLTOUCH
 
 /**
  * Touch-MI Probe by hotends.fr
@@ -1077,7 +1083,7 @@
  *
  * Specify a Probe position as { X, Y, Z }
  */
-#define NOZZLE_TO_PROBE_OFFSET { -35, -6, -0.5 }
+#define NOZZLE_TO_PROBE_OFFSET { -36, -4, -0.4 }
 
 // Most probes should stay away from the edges of the bed, but
 // with NOZZLE_AS_PROBE this can be negative for a wider probing area.
@@ -1126,7 +1132,7 @@
 #define Z_PROBE_LOW_POINT          -2 // Farthest distance below the trigger-point to go before stopping
 
 // For M851 give a range for adjusting the Z probe offset
-#define Z_PROBE_OFFSET_RANGE_MIN -20
+#define Z_PROBE_OFFSET_RANGE_MIN -50
 #define Z_PROBE_OFFSET_RANGE_MAX 20
 
 // Enable the M48 repeatability test to test probe accuracy
@@ -1355,7 +1361,7 @@
  */
 //#define AUTO_BED_LEVELING_3POINT
 //#define AUTO_BED_LEVELING_LINEAR
-// #define AUTO_BED_LEVELING_BILINEAR
+#define AUTO_BED_LEVELING_BILINEAR
 //#define AUTO_BED_LEVELING_UBL
 //#define MESH_BED_LEVELING
 
@@ -1508,16 +1514,16 @@
 // - Move the Z probe (or nozzle) to a defined XY point before Z Homing when homing all axes (G28).
 // - Prevent Z homing when the Z probe is outside bed area.
 //
-//#define Z_SAFE_HOMING
+#define Z_SAFE_HOMING
 
 #if ENABLED(Z_SAFE_HOMING)
   #define Z_SAFE_HOMING_X_POINT ((X_BED_SIZE) / 2)    // X point for Z homing when homing all axes (G28).
-  //#define Z_SAFE_HOMING_Y_POINT ((Y_BED_SIZE) / 2)  // Y point for Z homing when homing all axes (G28).
-  #define Z_SAFE_HOMING_Y_POINT 4
+  #define Z_SAFE_HOMING_Y_POINT ((Y_BED_SIZE) / 2)  // Y point for Z homing when homing all axes (G28).
+  //#define Z_SAFE_HOMING_Y_POINT 4
 #endif
 
 // Homing speeds (mm/m)
-#define HOMING_FEEDRATE_XY (40*60)
+#define HOMING_FEEDRATE_XY (50*60)
 #define HOMING_FEEDRATE_Z  (4*60)
 
 // Validate that endstops are triggered on homing moves
diff --git "a/C:\\Users\\Thomas\\Downloads\\Marlin2.0-lgt-support-LKxPro.orig\\Marlin\\Configuration_adv.h" "b/C:\\Users\\Thomas\\Downloads\\Marlin2.0-lgt-support-LKxPro\\Marlin\\Configuration_adv.h"
index 928c4bd..55beb5b 100644
--- "a/C:\\Users\\Thomas\\Downloads\\Marlin2.0-lgt-support-LKxPro.orig\\Marlin\\Configuration_adv.h"
+++ "b/C:\\Users\\Thomas\\Downloads\\Marlin2.0-lgt-support-LKxPro\\Marlin\\Configuration_adv.h"
@@ -1554,6 +1554,10 @@
   //#define MIN_PROBE_EDGE_RIGHT MIN_PROBE_EDGE
   //#define MIN_PROBE_EDGE_FRONT MIN_PROBE_EDGE
   //#define MIN_PROBE_EDGE_BACK MIN_PROBE_EDGE
+	#define MIN_PROBE_EDGE_LEFT  50
+	#define MIN_PROBE_EDGE_RIGHT 170
+	#define MIN_PROBE_EDGE_FRONT 50
+	#define MIN_PROBE_EDGE_BACK  170
 #endif
 
 #if EITHER(MESH_BED_LEVELING, AUTO_BED_LEVELING_UBL)

```

ThomasCr, 2021-09-28:
I got BLtouch bzw. 3dtouch pro running with the Marlin2.0-lgt-support-LKxPro branch... to bad that I got no response.
Development looks downsized/paused for a while.

It would be nice, when Longer would not develop there firmware out of tree (so it is hard for every body to update to a new/fresh version of marlin).

Marlin [officially supports LGT KIT V1.0 boards](https://github.com/MarlinFirmware/Marlin/pull/20372) and also have drivers to use the DGUS displays.

When Longer now start to push there patches directly to the marlin repository, users can just download the predefined config and run the compile button with a official checkout from the freshest marlin-2.0 firmware.

I have also tested the official marlin 2.0 sources, but without the display support - it was also running fine with my config.

closed 2021-09-28 by ThomasCr

LONGER3D, 2022-01-24:
https://github.com/LONGER3D/Marlin2.0-lgt/tree/LKxPro-bltouch-dualblower
There is LKxPro-bltouch-dualblower branch, please check it out

jacobsf, 2022-01-24:
Fantastic - thank you!

 
 

https://github.com/LONGER3D/Marlin2.0-lgt/tree/LKxPro-bltouch-dualblower