LONGER3D/Marlin2.0-lgt #11: [BUG] lib_deps TMC26XStepper not found

Ajtak, 2022-04-03:
Is not possible build with Visual Studio code, because TMC26XStepper library not found.

```
lib_deps          = ${common.lib_deps}
  TMC26XStepper=https://github.com/trinamic/TMC26XStepper/archive/master.zip
```

Error:
```
Library Manager: Installing https://github.com/trinamic/TMC26XStepper/archive/master.zip
PackageException: Got the unrecognized status code '404' when downloaded https://github.com/trinamic/TMC26XStepper/archive/master.zip
```

Clem69B, 2022-04-03:
Hello

Try with 
`  TMC26XStepper=https://github.com/interactive-matter/TMC26XStepper/archive/master.zip`

Seems to work for me

mhe-fr, 2022-05-17:
Hello,
The library is included in Marlin2.0-lgt\Marlin\src\module\stepper
So no need for this external library. 
To compile it I just comment the line.

```
[env:mega2560]
platform          = atmelavr
board             = megaatmega2560
board_build.f_cpu = 16000000L
lib_deps          = ${common.lib_deps}
#  TMC26XStepper=https://github.com/trinamic/TMC26XStepper/archive/master.zip
src_filter        = ${common.default_src_filter} +<src/HAL/AVR>

```