LONGER3D/Marlin2.0-lgt #14: [FR] Add documentation for CR Touch

isaaclepes, 2022-07-03:
### Description
The pinout for CR Touch probes is not well documented.  I wanted to share this information here, because it is the first place people are likely to look for this information.

### Feature Workflow

CR Touch Wiring

Servo Connection:
D7 = Yellow (Signal)
GND = White (Ground)
5v = Black (+5v)

Sensor Connection:
GND = Red (Ground)
5v = Blue (Z Min Stop Signal)

bob291, 2023-06-03:
I added a CR Touch to my Longer Lk5 Pro. I've done the wiring, designed and printed a mounting bracket. I am just not certain what I need to do next. I am completely stuck. I guess I thought it would be easier to find the information.

isaaclepes, 2023-06-03:
This may be helpful
https://www.reddit.com/r/3Dprinting/comments/wrxz5v/cr_touch_on_longer_lk_5_pro/