LONGER3D/Marlin2.0-lgt #16: Need source for DWTJB display. Techonics TJC4827X343_011_P03

jgbrown54, 2023-01-29:
Longer3d's current LK5 Pro and LK4 Pro use a Techonics display instead of the DWIN display. In order to add functionality, changes need to be made to the display firmware.

jgbrown54, 2023-07-02:
Does anyone read these issues? I've been waiting for months.

itsgudenuf, 2023-07-07:
Doesn't look like anyone read these.  
It looks like this screen might be a white label from Nexton? https://nextion.tech/
I need to get a closer look of the board but they have several models with the same 4827 part number, look similar and actually have dev files... I think Techonics is just a sales website/reseller as none of their links go to datasheets or the like, just a page to buy.